sparkfun/SparkFun_SCD30_Arduino_Library #1: Sparkfun_SCD30 library does not work on ESP8266

paulvha, 2018-08-12:
This  library does not work on an ESP8266 due to the way I2C Clockstretching is implemented in the ESP8266 library, in combination with the driver setup. 
Based on the on this library, I have created an extended version that does work on an ESP8266-Thing  + some extended features. It can be found on https://github.com/paulvha/scd30

nseidle, 2018-08-12:
Neat! I'd love to have ESPxx support. Please create a pull request and I'll get it pulled in.

paulvha, 2018-08-13:
I am not setup as a collaborator, I can not create a branch and pull request. I will drop you a PM showing the change to make.

paulvha, 2018-08-13:
I could not find your email. Attaching in a file the few changes needed. There are more changes that could be made (e.g. get serial number, better check on Temperature offset, running with ESP8266 as an access point or webserver). 
[SCD30_esp866-THING.txt](https://github.com/sparkfun/SparkFun_SCD30_Arduino_Library/files/2282861/SCD30_esp866-THING.txt)

nseidle, 2018-08-13:
You can create a PR from a fork at any time. 

    #if defined(ARDUINO_ESP8266_THING) || defined(ARDUINO_ESP8266_GENERIC) || defined(ARDUINO_ESP8266_ESP01) || defined(ARDUINO_ESP8266_ESP13) || defined(ARDUINO_ESP8266_ESP12) || defined(ARDUINO_ESP8266_NODEMCU) || defined(ARDUINO_ESP8266_THING_DEV) || defined (ARDUINO_ESP8266_ESP210) || defined(ARDUINO_MOD_WIFI_ESP8266) || defined(ARDUINO_ESP8266_PHOENIX_V1) || defined(ARDUINO_ESP8266_PHOENIX_V2) || defined(ARDUINO_ESP8266_ARDUINO)
        _i2cPort->setClockStretchLimit(200000);
    #endif

Can you use the ARDUINO_ARCH_ESP32 and ARDUINO_ARCH_ESP8266 tags instead?

This is really helpful. Once you have a PR I'll test on a few platforms.

paulvha, 2018-08-13:
completed the PR... look forward to your results.

closed 2018-09-12 by LGKev