sparkfun/SparkFun_SCD30_Arduino_Library #3: Set Forced Recalibration value (FRC)

Sedude, 2018-11-23:
Hi, I'm using the SCD30 sensor for my masterthesis. Therefore I need to calibrate it. Sadly I could not find the command for setting the Forced Recalibration value (FRC) in the library. Does anybody know how to implement the given example in the "Interface Description Sensirion SCD30 Sensor Module" sheet from Sensirion into the library? Thanks a lot!
[SCD30_Interface_Description.pdf](https://github.com/sparkfun/SparkFun_SCD30_Arduino_Library/files/2610896/SCD30_Interface_Description.pdf)

mut0g3n, 2020-03-05:
Hi, Just found this thread. I cannot get the FRC function to work. SCD30 otherwise works fine connected to my Adafruit Huzzah Feather. Tried using the FRC function in the library but when writing it into my code in the Arduino app (Mac OS X 10.14), the code line does not change text colour indicadting a recognised function. Tried pointing at the library directly "SCD30::" but same problem. By contrast the ARC function does change colour. Bit of a newbie with Arduino but will post a screen shot and code later. Any suggestions?

alfkoehn, 2020-11-16:
I just want to add a reply here because this page was ranked quite high in my search for some information about the forced recalibration: the command to do this is `setForcedRecalibrationFactor`, if that was the question. 

According to the datasheet, the sensor should be placed in stable conditions for at least 2 minutes before applying the forced recalibration. As a value, I used 410, as this is roughly the value we have outside at fresh air (well, at least at the moment...).