sparkfun/SparkFun_SCD30_Arduino_Library #6: readRegister always returns 0

tayRpi, 2019-05-26:
I have a SCD30 sensor, however `dataAvailable()` is always `false`. 

I debugged this:  `dataAvailable()` calls `uint16_t SCD30::readRegister(uint16_t registerAddress)`, which has the following strange behaviour: When calling the function, `_i2cPort->available()` is `2`, so the msb and lsb are assembled and returned:
```
    uint8_t msb = _i2cPort->read();
    uint8_t lsb = _i2cPort->read();
    return ((uint16_t)msb << 8 | lsb);
```
The returned value is `0`. Any idea what is wrong?

bobobo1618, 2019-09-01:
This indicates that the sensor is functioning and responding correctly but doesn't have any data available for you to read.

Can you post your full code somewhere? Have you called `begin()` somewhere in your code's setup?

nseidle, 2020-01-22:
Closing due to inactivity. If you still have problems please consider giving us more info: What platform are you using? Can you post the code to replicate the problem?

closed 2020-01-22 by nseidle