sparkfun/SparkFun_SCD30_Arduino_Library #9: Constant spikes in measurements

CellDope, 2020-02-17:
I used this driver with ESP32. Overall it works fine but I get constant spikes for Co2 levels. Spikes seem to have quite constant pattern (seasonality). 
Humidity and Temperature from other hand does not have any spikes and seems to work flawless.

Does anyone have a good guess why this is happening? I have same issue with automatic calibration enabled and disabled. Please let me know if more data is required.

data: https://pastebin.com/c3yLYiAX
chart: https://pasteboard.co/IV6fGFW.png 

Thanks!

CellDope, 2020-02-17:
Just to clarify, wanted to add that I'm reporting about random spikes where one measurement all of a sudden is higher than previous reading and next reading instantly goes back to normal. 
I've highlighted readings when sensor was at home with no one in the room in this screenshot:
https://pasteboard.co/IV6PNPH.png

When we are looking on the reading with people near-by its noticeable that sensor have the same behavior but the seasonal spikes are less noticeable because they are mixed with real readings.

nseidle, 2020-02-24:
I really can't explain or dig into how the sensor gathers its readings. This is really a question for [Sensirion](https://github.com/Sensirion).

closed 2020-02-24 by nseidle