sparkfun/SparkFun_SCD30_Arduino_Library #11: (FRC) Forced recalibration function.

mut0g3n, 2020-03-05:
Hi, I cannot get the FRC function to work. SCD30 otherwise works fine connected to my Adafruit Huzzah Feather with good data output. Tried however using the FRC function in the library but when writing it into my code in the Arduino app (Mac OS X 10.14), the code line does not change text colour indicating a recognised function. Also tried pointing at the library directly "SCD30::" but same problem. By contrast the ARC function does change colour. Bit of a newbie with Arduino but will post a screen shot and code later but code is just the first example with the addition of the FRC function included in the setup.  Any suggestions?
 (Listing as new issue becuse prior FRC issue was closed, sorry if that is the wrong way. )

jobr97, 2020-03-17:
How excatly are you trying to call the function? Can you paste a code snippet? The name of the function is `setForcedRecalibrationFactor()`

LGKev, 2020-03-17:
Hi there!

I was wondering if possible to remove my LGKev github account from SparkFun/ SparkX activities. I keep getting emails and I think my account is linked to Sparkfun’s libraries 

Thanks!
Hope you are all doing well :D

jobr97, 2020-03-17:
@LGKev you are watching the repo. I think that's why you get notifications...

AndyEngland521, 2020-03-17:
Yeah @LGKev I don't think I have the ability to block you from the Repo, just go ahead and unwatch in the future, sorry for the inconvenience..

@mut0g3n It looks like, as @jobr97 said, it's not in the keywords.txt, which is how the IDE knows which text strings to color. In the future, this sort of thing won't break your code and you could still compile and things would functionally be the same, but filing an issue is definitely the proper course of action, as we like our code to look good as well as function well. Thanks! This should be added in v1.0.7

closed 2020-03-17 by AndyEngland521

mut0g3n, 2020-03-19:
Very many thanks @jobr97 and @AndyEngland521!

v1.0.7 now working perfectly.