sparkfun/SparkFun_SCD30_Arduino_Library #16: Faulty CRC computation

MBjoern, 2020-11-13:
The interface for `setForcedRecalibrationFactor` is not working.
I assume that the CRC computation is faulty, as I'd expect the checksum to be 0x50, but the code here is getting a 0x33.
Hence the forced recalibration is never executed on the chip.

Surprisingly, the other checksums are correct, only this one seams to be affected.

nseidle, 2020-12-31:
Do you have some simple code that can demonstrate your issue?

awatterott, 2020-12-31:
I can confirm that the `setForcedRecalibrationFactor()` is working. Maybe the issue is related when the function is called to fast after power-up. The datasheet says that at least 2 minutes in a stable environment are needed and from our experiences it is not working or has no effect, when the time is too short.

PaulZC, 2021-02-01:
Hi @MBjoern ,
Are you still having problems with ```setForcedRecalibrationFactor```?
Best wishes,
Paul

MBjoern, 2021-02-01:
Hi Paul

Thank you for getting back to me. I did not manage to further test the issue. I've just noticed the discrepancy w.r.t. the CRC value.
If the library works for others, then it appears to be a problem on my end. This issue can be closed accordingly. Thank you.

closed 2021-02-01 by MBjoern