sparkfun/SparkFun_SCD30_Arduino_Library #26: getMeasurementInterval() is not implemented!

physikant, 2021-09-01:
I noticed that the getMeasurementInterval() function is not implemented.
I'm not an experienced C programmer, but I managed to implement it myself by adding
"uint16_t getMeasurementInterval(void); "
on line 96 of the header file and adding
"//Get the measurement interval. See 1.3.8.
uint16_t SCD30::getMeasurementInterval(void)
{
  return readRegister(COMMAND_SET_MEASUREMENT_INTERVAL);
}"
to the cpp file.

nseidle, 2021-09-01:
[It is](https://github.com/sparkfun/SparkFun_SCD30_Arduino_Library/blob/main/src/SparkFun_SCD30_Arduino_Library.h#L87) already implemented, but a helper function would be nice. I'll see what I can do.

nseidle, 2021-09-01:
Checkout ex 2 in the latest release. Please let us know if you have problems.

physikant, 2021-09-01:
Great, it is working for me. Thanks a lot!

nseidle, 2021-09-02:
Good to hear!