sparkfun/SparkFun_SCD30_Arduino_Library #27: Temperature offset not working correctly

patfelst, 2021-11-12:
Hi there, thanks for this library.

I'm using the latest version (1.0.14) on a M5 Stack Core2 (ESP32) and things are working ok, except for the temperature offset. When I set a positive offset of, there is no change to the temperature reading. If I set a negative offset, I get incorrect temperature readings and it affects the humidity too. Then setting the offset back to zero, the temperature then slowly begins to climb back to its correct value. Here is the test code I used to demonstrate this.

Note there is a bug in the function `.getTemperatureOffset()` where it should convert the offset read back from the sensor to a signed integer, not unsigned. I don't think this is why the offset isn't working though.

Here is a small program that demonstrates what I'm seeing:

```C++
void scd30_temp_offset(float temp_offs) {
  airSensor.setTemperatureOffset(temp_offs);
  while (!airSensor.dataAvailable());
  Serial.printf("Set temp offset = %.2f, Temperature %.2f, get temp offset %.3f\n",
                temp_offs, airSensor.getTemperature(), airSensor.getTemperatureOffset());
}

void scd30_read_temp(void) {
  while (!airSensor.dataAvailable());
  Serial.printf("Temperature %.2f, get temp offset %.3f\n", airSensor.getTemperature(), airSensor.getTemperatureOffset());
}

void setup(void) {
  M5.begin(true, false, true, true, false);  // Start M5 Stack Core2
  M5.Axp.SetLed(0);                          // Turn off green LED

  Wire.begin(32, 33);  // Start I2C for CO2 sensor. Using I2C pins SDA = 32, SCL = 33
  airSensor.begin();   // Start Sensirion SCD-30 CO2 sensor
  airSensor.reset();
  delay(100);
  scd30_temp_offset(0.0);
  scd30_temp_offset(1.0);
  scd30_temp_offset(2.0);
  scd30_temp_offset(3.0);
  scd30_temp_offset(2.0);
  scd30_temp_offset(1.0);
  scd30_temp_offset(0.5);
  scd30_temp_offset(0.0);
  scd30_temp_offset(-0.5);
  scd30_temp_offset(-1.0);
  scd30_temp_offset(-2.0);
  scd30_temp_offset(-3.0);
  scd30_temp_offset(0.0);

  for (uint16_t i = 0; i < 15; i++) {
    scd30_read_temp();
  }
}
```

The serial output from the above is:

```
Set temp offset = 0.00, Temperature 24.82, get temp offset 0.000
Set temp offset = 1.00, Temperature 24.81, get temp offset 1.000
Set temp offset = 2.00, Temperature 24.75, get temp offset 2.000
Set temp offset = 3.00, Temperature 24.73, get temp offset 3.000
Set temp offset = 2.00, Temperature 24.69, get temp offset 2.000
Set temp offset = 1.00, Temperature 24.67, get temp offset 1.000
Set temp offset = 0.50, Temperature 24.65, get temp offset 0.500
Set temp offset = 0.00, Temperature 24.68, get temp offset 0.000
Set temp offset = -0.50, Temperature 17.53, get temp offset 654.860
Set temp offset = -1.00, Temperature 10.40, get temp offset 654.360
Set temp offset = -2.00, Temperature 3.47, get temp offset 653.360
Set temp offset = -3.00, Temperature -3.46, get temp offset 652.360
Set temp offset = 0.00, Temperature -3.20, get temp offset 0.000
Temperature -2.88, get temp offset 0.000
Temperature -2.57, get temp offset 0.000
Temperature -2.26, get temp offset 0.000
Temperature -1.96, get temp offset 0.000
Temperature -1.68, get temp offset 0.000
Temperature -1.36, get temp offset 0.000
Temperature -1.02, get temp offset 0.000
Temperature -0.75, get temp offset 0.000
Temperature -0.50, get temp offset 0.000
Temperature -0.19, get temp offset 0.000
Temperature 0.09, get temp offset 0.000
Temperature 0.37, get temp offset 0.000
Temperature 0.61, get temp offset 0.000
Temperature 0.91, get temp offset 0.000
Temperature 1.25, get temp offset 0.000
```

To fix the bug in function `.getTemperatureOffset()` the `uint16_t` response needs to be `int16_t`
```C++
//Get the temperature offset. See 1.3.8.
float SCD30::getTemperatureOffset(void)
{
  // uint16_t response = readRegister(COMMAND_SET_TEMPERATURE_OFFSET);
  int16_t response = readRegister(COMMAND_SET_TEMPERATURE_OFFSET);
  return (((float)response) / 100.0);
}
```

and the output is then:
```
Set temp offset = 0.00, Temperature 24.79, get temp offset 0.000
Set temp offset = 1.00, Temperature 24.77, get temp offset 1.000
Set temp offset = 2.00, Temperature 24.73, get temp offset 2.000
Set temp offset = 3.00, Temperature 24.68, get temp offset 3.000
Set temp offset = 2.00, Temperature 24.66, get temp offset 2.000
Set temp offset = 1.00, Temperature 24.67, get temp offset 1.000
Set temp offset = 0.50, Temperature 24.66, get temp offset 0.500
Set temp offset = 0.00, Temperature 24.66, get temp offset 0.000
Set temp offset = -0.50, Temperature 17.50, get temp offset -0.500
Set temp offset = -1.00, Temperature 10.37, get temp offset -1.000
Set temp offset = -2.00, Temperature 3.37, get temp offset -2.000
Set temp offset = -3.00, Temperature -3.57, get temp offset -3.000
Set temp offset = 0.00, Temperature -3.26, get temp offset 0.000
Temperature -2.93, get temp offset 0.000
Temperature -2.63, get temp offset 0.000
Temperature -2.30, get temp offset 0.000
Temperature -1.99, get temp offset 0.000
Temperature -1.70, get temp offset 0.000
Temperature -1.40, get temp offset 0.000
Temperature -1.12, get temp offset 0.000
Temperature -0.82, get temp offset 0.000
Temperature -0.54, get temp offset 0.000
Temperature -0.25, get temp offset 0.000
Temperature 0.07, get temp offset 0.000
Temperature 0.36, get temp offset 0.000
Temperature 0.66, get temp offset 0.000
Temperature 0.91, get temp offset 0.000
Temperature 1.19, get temp offset 0.000
```
But the temperature offset is still not working.

nseidle, 2021-11-16:
Good bug find on the getTemperatureOffset() not properly returning negative values. That will be fixed momentarily.

I can replicate the weird getTemperature() readings on an Uno. It's a bit confusing because [Sensirion](https://cdn.sparkfun.com/assets/f/e/d/0/1/Sensirion_CO2_Sensors_SCD30_Interface_Description.pdf) says temp offset is a uint16.

When writing a positive offset (say 3.2), power cycling the sensor shortens the 'return to normal' window to basically normal at start. It looks like their algorithm assumes the temp sensor is outputting correct data at POR and then begins trying to take the offset into account.

When writing a negative offset (say -2.2) power cycling appears to cause the sensor to start at room temp then slowly makes it way to -650C... Wacky. 
[image]

I really don't know. I'll reach out to the few contacts I have at Sensirion to see if they have any info.

nseidle, 2021-11-17:
Ah, with a little help from Sensirion, see the [Low Power Mode](https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/9.5_CO2/Sensirion_CO2_Sensors_SCD30_Low_Power_Mode.pdf) doc.

> the SCD30 offset temperature is obtained by subtracting the reference temperature from the SCD30 output temperature

[image]

Therefore, as I understand it, the setTemperatureOffset value is *subtracted* from the sensor's measured temperature.

Along those line, my contact says the device uses a heater during detection and as such, it was never thought the sensor would report a value too low, only too high. Therefore, the offset (which is subtracted) can only be positive.

Clear as mud?

patfelst, 2021-11-17:
Hi Nate, thanks for looking into this. It actually does make sense as with a heater the temp will always be too high until it is offset. Probably simpler to set T_old_offset to zero first, then the equation becomes simpler.

nseidle, 2021-11-18:
Np. Notes added to lib. setTempOffset now only accepts >0 values.

closed 2021-11-18 by nseidle