sparkfun/SparkFun_SCD30_Arduino_Library #29: When the sensor is disconnected the CO2 value persist

hpsaturn, 2022-01-28:
## Summary

I'm working on a wrapper of multiples Air quality sensors, this wrapper has sensors and units with dynamically registers, for this reason it's important that the values when the sensor doesn't exists more going to reset.

## steps to replicate the issue

1. disconnect the sensor
2. `uint16_t tCO2 = scd30.getCO2();` persist

nseidle, 2022-01-28:
Hmm. We are [not testing](https://github.com/sparkfun/SparkFun_SCD30_Arduino_Library/blob/main/src/SparkFun_SCD30_Arduino_Library.cpp#L99) if the measurement was successful. 

Ugh. And we don't have an isConnected function. Who [wrote this library](https://github.com/sparkfun/SparkFun_SCD30_Arduino_Library/commit/1c3f9332a5af85a6e5859b55b909fd81d2bc3110)!?

nseidle, 2022-01-28:
Thanks for reporting. I'll try to get this fixed shortly.

nseidle, 2022-01-28:
Ok - I've added an isConnected() function for you so you should be able to ping the sensor to see if it's there/alive or disconnected.

Also, co2, humidity, and temperature should all report 0 upon sensor reading failure. Not the greatest error checking but as long as you're checking isConnected you should be ok.

closed 2022-01-28 by nseidle