sparkfun/SparkFun_SCD30_Arduino_Library #32: Connecting SCD30 to Arduino IOT Cloud

Pedrocuberos, 2022-03-18:
I am trying to get readings from the board and when doing it on IDE, I can perfectly read and print the values. However, when trying to do the same on the Arduino IoT Cloud the board starts connecting and disconnecting from the USB port. At the same time, the board loses connectivity to WiFi so I have to restore the board before trying to load code again. 

`/*
  float cO2;
  float humidity;
  float temp;
*/

#include "thingProperties.h"
#include <Wire.h>
#include <SparkFun_SCD30_Arduino_Library.h>

SCD30 airSensor;

void setup() {

  Serial.begin(115200);
  Wire.begin();

  initProperties();
  ArduinoCloud.begin(ArduinoIoTPreferredConnection);
  setDebugMessageLevel(2);
  ArduinoCloud.printDebugInfo();

}

void loop() {

  ArduinoCloud.update();

 if (airSensor.dataAvailable()) {

    cO2 = airSensor.getCO2();
    humidity = airSensor.getTemperature();
    temp = airSensor.getHumidity();

    delay(500);

    } else {

    delay(500);

    }

}`

And this is the piece of code that creates the issue

`if (airSensor.dataAvailable()) {

    cO2 = airSensor.getCO2();
    humidity = airSensor.getTemperature();
    temp = airSensor.getHumidity();

    delay(500);

    } else {

    delay(500);

    }`

Could you please help me? I am losing my mind here.

nseidle, 2022-03-18:
What board do you have the SCD30 connected to?

Pedrocuberos, 2022-03-18:
I am using the Arduino Uno Nano 33 IoT

Pedrocuberos, 2022-03-18:
>

Pedrocuberos, 2022-03-18:
I apologize, I closed the issue by mistake... I'm new to this.

nseidle, 2022-03-18:
Unfortunately I don't have one of those on hand to test at the moment. 

To be clear, you can run the example 1 without issue? But board resets when you turn on WiFi? Sounds like a power issue. 

The SCD30 itself will draw ~75mA during measurement:

[image]

This is not a library issue. I suspect the Arduino BLE board is browning out when the NINA turns on to transmit. I recommend making sure your power supply can tolerate the surges in current, then measuring the board's VCC over time to verify the 3.3V supply is not dropping. Sorry, I can't be more help.

closed 2022-03-18 by nseidle

Pedrocuberos, 2022-03-18:
Thanks for the support. I am not clear why this is a power issue if I see other people have successfully connected this to the Blynk app and server using the same wiring configuration and powering this from a USB port